UbiquityRobotics/fiducials #263: /fiducial_transform not publishing when using USB camera

roaz1801, 2021-12-09:
Hello, I'm trying to get the transform of a marker that has been detected. Using ROS Noetic, a USB camera with the cv_camera package, and the camera is attached to a turtlebot. Using a USB camera because I couldn't get the raspi cam to work. 

The USB camera works and is detecting a marker when checking the rqt image, and the when echoing the /fiducial_vertices topic you get feedback from it. The /fiducial_transform however does not print anything when using echo, no fiducial shows up in the tf_tree. This all worked well in the simulations, but now after moving to hardware and changing to a USB camera it is not working. 

This is the launch file for the turtlebot:

```
<launch>
   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_robot.launch" ns="tb3_0">
  </include>

  <arg name="model" default="waffle_pi"/>
  <arg name="first_tb3" default="tb3_0"/>

  <node pkg = "cv_camera" type="cv_camera_node" name="cv_camera" output="screen">
   <param name="image_width" value="640"/>
   <param name="image_heigth" value="480"/>
   <param name="rate" value="60"/>
   </node>

  <node pkg="tf" type="static_transform_publisher" name="camera_frames_pub" args="0.05 0.0 0.1 0 0 0 tb3_0/base_link /camera 35"/>

     <!-- Start marker detector-->
 <node name="aruco_detect" pkg="aruco_detect" type="aruco_detect">
    <param name="image_transport" value="compressed" />
    <param name="publish_images" value="true" />
    <param name="fiducial_len" value="0.15" />
    <param name="dictionary" value="0" />
    <param name="do_pose_estimation" value="true" />
    <param name="ignore_fiducials" value="" />
    <param name="fiducial_len_override" value="" />
    <remap from="/camera/compressed" to="/cv_camera/image_raw/compressed"/>
    <remap from="/camera_info" to="/cv_camera/camera_info"/>
  </node>

 <group ns = "$(arg first_tb3)">
    <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find turtlebot3_description)/urdf/turtlebot3_$(arg model).urdf.xacro' botname:=$(arg firs>

     <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher" output="screen">
      <param name="publish_frequency" type="double" value="50.0" />
      <param name="tf_prefix" value="$(arg first_tb3)" />
    </node>
 </group>
 </launch>
```

where the main difference between this and the launch file used in the simulations is the remapping of the camera/compressed and caemra/info. 

Also asked the same question here: https://answers.ros.org/question/392521/aruco_detect-fiducial_transform-not-giving-any-transforms-when-using-usb-camera/

Thanks in advance

anfederman, 2021-12-09:
I have gotten Aruco detect to work on a USB cam in Noetic. I think you are
correct in thinking it is a transform error. I will spin up a USB camera
and send you my launch files.

roaz1801, 2021-12-09:
I did try making the rospicam work on noetic but at some point I decided it was just too much effort when a USB camera could work too. Tried to make it work with both the image given in the https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup
and the default image from Raspberry Pi Imager. Never got raspistill to work there.

anfederman, 2021-12-09:
Try:

Add camera support libraries and test:

sudo apt install libraspberrypi-bin libraspberrypi-dev

this worked on ARM64 Noetic  Pi 4 Ubuntu 20.04 built from the server image.

   then raspistill -o test.jpg

roaz1801, 2021-12-09:
> Try: Add camera support libraries and test: sudo apt install libraspberrypi-bin libraspberrypi-dev this worked on ARM64 Noetic Pi 4 Ubuntu 20.04 built from the server image. then raspistill -o test.jpg
> […](#)

Thank you, will try this next time I get to use the physical turtlebot and report back, but it might take a few days until next time I get to use the hardware.

roaz1801, 2021-12-09:
Also, I have now tested using only a USB camera not connected to anything, so I don't think this issue has anything to do with the turtlebot at all, might just be that the aruco_detect node is not writing to the right topic. From the /fiducial_vertices I get:

header: 
  seq: 6092
  stamp: 
    secs: 1639076096
    nsecs: 909698461
  frame_id: ''
image_seq: 6093
fiducials: 
  - 
    fiducial_id: 0
    direction: 0
    x0: 499.4951171875
    y0: 222.32830810546875
    x1: 552.80908203125
    y1: 223.65765380859375
    x2: 549.2293701171875
    y2: 277.81787109375
    x3: 495.5509033203125
    y3: 274.6446228027344
---
And in the rqt image viewer the marker is being detected.

anfederman, 2021-12-09:
This is the aruco launch filefor raspicam on a Magni robot:

<launch>
  <arg name="mapping_mode" default="false"/>
  <arg name="fiducial_len" default="0.140"/>

  
  <include file="$(find
raspicam_node)/launch/camerav2_1280x960_10fps.launch" />

  
  <include file="$(find aruco_detect)/launch/aruco_detect.launch">
    <arg name="camera" value="/raspicam_node"/>
    <arg name="image" value="image"/>
    <arg name="transport" value="compressed"/>

    
    <arg name="fiducial_len" value="$(arg fiducial_len)" />
  </include>

  
  <include file="$(find fiducial_slam)/launch/fiducial_slam.launch">
    <arg name="map_frame" value="map"/>
    <arg name="odom_frame" value="odom"/>
    <arg name="base_frame" value="base_footprint"/>
    <arg name="future_date_transforms" value="0.5"/>
  </include>

-----

If it works,in rqt_image_view  you should see both camera and fiducial
image topics.

roaz1801, 2021-12-09:
Is there no way of making this work without a raspicam/ raspicam_node? Would like to use it it with a USB camera

anfederman, 2021-12-09:
I had it working with an Xtion (3d camera, using the camera_rgb image (this
is via USB)in both Kinetic and Noetic.  I am not positive,
but I think I had it work both on a laptop and a Jetson.  I think you will
need to have a calibrated camera however.

anfederman, 2021-12-09:
in the launch file I sent

change the camera by  the include file to find  "usb_cam"  (or whatever the
camera node is.

in the fiducial detection section change the camera value.

roaz1801, 2021-12-09:
This is what I've done with the launch code. The cv_camera has no launch file so the node is directly put in, and the camera value has been changed. Not sure if this is what the camera value should be changed into, but I don't know what else it could be.
```
<launch>
  <arg name="mapping_mode" default="false"/>
  <arg name="fiducial_len" default="0.140"/>

  <!-- camera -->
  <node pkg = "cv_camera" type="cv_camera_node" name="cv_camera" output="screen">
   <param name="image_width" value="640"/>
   <param name="image_heigth" value="480"/>
   </node>

  <!-- Fiducial detection -->
  <include file="$(find aruco_detect)/launch/aruco_detect.launch">
    <arg name="camera" value="/cv_camera"/>
    <arg name="image" value="image"/>
    <arg name="transport" value="compressed"/>

    <!-- fiducial_len -->
    <arg name="fiducial_len" value="$(arg fiducial_len)" />
  </include>

  <!-- Fiducial slam -->
  <include file="$(find fiducial_slam)/launch/fiducial_slam.launch">
    <arg name="map_frame" value="map"/>
    <arg name="odom_frame" value="odom"/>
    <arg name="base_frame" value="base_footprint"/>
    <arg name="future_date_transforms" value="0.5"/>
  </include>
  </launch>
```

anfederman, 2021-12-09:
So rostopic list

Rostopic echo / camera image topic
                           / "Fiducial_image topic'

See if it works, then view it with rqt_image_view.

If you can list the topic, but not see when echoing, the issue is
networking, most likely how hostnames and IPs are specified.

roaz1801, 2021-12-09:
Alright, let me give all the info I can:
rostopic list:

```
/aruco_detect/compressed/parameter_descriptions
/aruco_detect/compressed/parameter_updates
/aruco_detect/parameter_descriptions
/aruco_detect/parameter_updates
/cv_camera/camera_info
/cv_camera/image_raw
/cv_camera/image_raw/compressed
/cv_camera/image_raw/compressed/parameter_descriptions
/cv_camera/image_raw/compressed/parameter_updates
/cv_camera/image_raw/compressedDepth
/cv_camera/image_raw/compressedDepth/parameter_descriptions
/cv_camera/image_raw/compressedDepth/parameter_updates
/cv_camera/image_raw/theora
/cv_camera/image_raw/theora/parameter_descriptions
/cv_camera/image_raw/theora/parameter_updates
/fiducial_images
/fiducial_images/compressed
/fiducial_images/compressed/parameter_descriptions
/fiducial_images/compressed/parameter_updates
/fiducial_images/compressedDepth
/fiducial_images/compressedDepth/parameter_descriptions
/fiducial_images/compressedDepth/parameter_updates
/fiducial_images/theora
/fiducial_images/theora/parameter_descriptions
/fiducial_images/theora/parameter_updates
/fiducial_map
/fiducial_pose
/fiducial_slam/camera_pose
/fiducial_transforms
/fiducial_vertices
/fiducials
/ignore_fiducials
/rosout
/rosout_agg
/tf
/tf_static

```
then rostopic echo the /fiducial_images, something is echoed out (big arrays of numbers).
rqt_image_view shows /fiducial_image. The marker is even being detected, as can be seen here:
[image]

The issue comes now, as the problem is that the /fiducial_transforms is not working as it should. Even though the marker is being detected, and the vertices are being detected, the pose of the marker in relation to the camera is not coming out, as can be seen on the image too in the tf_tree

anfederman, 2021-12-09:
Ok so the issue is ROS doesn't know where the camera is in relation to the
robot.  This is usually fixed by linking the camera to the base_link or
base_fpotprint of the robot. In RVIZ when you look at the topics being
displayed you will see things flagged as red with warning messages about
can't display because no transform exists.. blah blah.

This is fixed by adding a static transform in the launch, parameters
typically in robot bringup or description.

If you download Ubiquity Robotics GitHub for Magni robot, you'll see where
we can specify different positions for the camera so fiducial SLAM works.
I believe the robot.yaml, bring up and description sections all have pieces
of the needed code.

PS everything about robotics is hard. There is no learning curve for ROS.
There is a learning cliff. Hold your nose and jump.

roaz1801, 2021-12-09:
I did have this line in the first launch file:
` <node pkg="tf" type="static_transform_publisher" name="camera_frames_pub" args="0.05 0.0 0.1 0 0 0 tb3_0/base_link /camera 35"/>
`
So I don't think it's a problem with the camera not knowing where it is in relation to the robot, I still think there's something that's making it so the aruco_detect does not manage to get the frame_id from the cv_camera/camera_info topic.

But anyway, I'll look more into it tomorrow. Thank you for your time!

anfederman, 2021-12-10:
Fiducial SLAM works by assuming markers are fixed, and the camera is
attached to the robot. For it to work, the position of the camera and how
it is pointed, relative to the base_link of the robot is used to compute
the motion of the robot based on transforms to odom, cmd_vel, and map
topics. This further gets confused when we add scan topics from lasers and
obstacle avoidance from sonar.

The entire reason Ubiquity Robotics developed fiducial SLAM was it was
inexpensive and Lidar's 10 years ago were twice the cost of the robot. Now
there are decent time of flight lidar packages for $130, and 3d cameras for
similar prices, we are moving away from fiducial SLAM, and using cv cameras
for informational navigation markers.

roaz1801, 2021-12-10:
> I had it working with an Xtion (3d camera, using the camera_rgb image (this is via USB)in both Kinetic and Noetic. I am not positive, but I think I had it work both on a laptop and a Jetson. I think you will need to have a calibrated camera however.
> […](#)

This seems to be where the issue was. My camera was not calibrated and I managed to calibrate it today. After having done this, the /fiducial_transform works as it should. Thank you for mentioning that.

closed 2021-12-10 by roaz1801