UbiquityRobotics/fiducials #264: No detection near the marker

xperiazu21, 2022-01-20:
Hello, when the camera is about 40 cm away from the marker, the ArUco marker is no longer detected.

Seeing that #191 has the same problem, I changed the maxMarkerPerimeterRate parameter in aruco_detect through rqt_reconfigure, the default value is 1.0, I lowered the value, but the distance of the marker can be detected further.

I also tried to replace the markers of other dictionaries, the result is the same, are there any parameters I need to adjust?
[image]
[image]

MoffKalast, 2022-01-20:
Yeah I've been seeing this a lot, though it seems to only happen at certain times. It does happen if the dictionary isn't set up right, though it also happens sometimes when it is.

xperiazu21, 2022-01-21:
I saw a few pull requests about maxMarkerPerimeterRate.

Need to modify file aruco_detect/cfg/DetectorParams.cfg 
`gen.add("maxMarkerPerimeterRate",                 double_t, 0,
        "Determine maximum perimeter for marker contour to be detected. This is defined as a rate respect to the maximum dimension of the input image",
        4.0, 0, 1)`

**change 1 to 4.0**
`gen.add("maxMarkerPerimeterRate",                 double_t, 0,
        "Determine maximum perimeter for marker contour to be detected. This is defined as a rate respect to the maximum dimension of the input image",
        4.0, 0, 4.0)`

**After the change, the catkin_make must be executed again to take effect!**
 I am stuck for a long time.

closed 2022-01-21 by xperiazu21