UbiquityRobotics/fiducials #266: Not able to visualize /fiducial_images via image_view with usb_cam

Trushantadeshara, 2022-02-22:
I am using usb_cam node and with that I want to use image_view but even after remapping I am not able to visualize it.
I remapped /camera_info to /usb_cam/camera_info and /camera to /usb_cam/image_raw
I am able to detect ArUCo markers

Trushantadeshara, 2022-02-22:
I was able to solve it by remapping /camera/compressed to /usb_cam/image_raw/compressed

Trushantadeshara, 2022-02-22:
The thing is when I run launch file I am able to echo on topics like /fiducial_images and /fiducial_images/compressed but same is not happening when I perform rosrun with same parameters.

MoffKalast, 2022-02-22:
In that case I suggest you use roslaunch and set up the correct topic remaps as you've already figured out. It defaults to raspicam and it's not intended to automatically detect all image topics on the system, only web streaming nodes have that capability out of any image processing nodes I've ever seen.

Trushantadeshara, 2022-02-22:
Ohh cool, I got it. One more thing that you said about default to raspicam, that is in aruco_detect launch file? Because I am trying to directly run aruco node

anfederman, 2022-02-22:
Can you be more specific?

Are you launching and running the usb_cam node on a robot or remote , and
rostopic echo and rosrun image_view on a workstation?

On the workstation, do rostopic list and rostopic echo show the topics and
video data, but image_view doesn't show anything? If the data is missing or
image view doesn't work on the workstation it might mean the ROS_MASTER_URI
isn't set correctly.

If echo works, but image-view doesn't, I don't know what the problem could
be, perhaps a networking issue. What happens when you run everything from
the workstation? I would try to get that working first. Just ROS, usb_cam
and image_view on one platform. If that works for you, let me know.

MoffKalast, 2022-02-22:
> One more thing that you said about default to raspicam, that is in aruco_detect launch file? Because I am trying to directly run aruco node

Ah my bad, it seems to actually be just "camera" by default, odd. https://github.com/UbiquityRobotics/fiducials/blob/1ae8a13d56fafa436aca181da451c1bf808a288c/aruco_detect/src/aruco_detect.cpp#L673

I'm not sure why you'd want to run it via rosrun though, wouldn't it make more sense to copy paste the existing launch file and set it up the way your system needs it to be?

anfederman, 2022-02-22:
The default aruco_detect in magni_nav launches raspicam_node I think at 420
resolution. I think it would use the Magni default forward raspicam URDF. I
think in breadcrumb, the camera gets flipped 180.

Trushantadeshara, 2022-02-23:
I am running everything directly on the robot and I am using Jetson Nano for this. The thing is I was able to run everything usb_cam, aruco_detect and image_view with launch file but I wish to know the reason why I am not able to visualize in image_view when I rosrun all of them individually considering all the parameters are same.

anfederman, 2022-02-23:
Which version of ubuntu  and ROS? I am pretty sure it has worked on a Nano
using an Asus depth CAM and Melodic. Never tested with Noetic. Also not
image_view, but rqt_image_view.

Trushantadeshara, 2022-02-23:
Ubuntu 18.04 with ROS Melodic

Trushantadeshara, 2022-03-11:
I think I found the issue, due to some reason display for Jetson Nano is acting different which might have caused error in taking long command inputs (extra spaces, missing character). I was able to solve this by using a launch file. Closing this issue.

closed 2022-03-11 by Trushantadeshara