UbiquityRobotics/fiducials #268: Extreme pose jitter and map doesn't match

Wayne-xixi, 2022-02-24:
Hi, I use the fiducials in gazebo Simulation. I first placed about 30 fiducials (14cm long) on the wall of a square room in gazebo's world. And I used a mobile robot with a camera to detect the fiducials to estimate the pose of the robot. The scene is as follows:
[image]
So far I have mainly encountered two problems:
1. The map doesn't match the real fiducials in gazebo world. All of the fiducials should be at a height of 0.3 -0.4m, but the map from fiducial_slam is very different from gazebo world. And it seemed that the fiducials are detected in the wrong poses.
[image]

2.  Since I struggled for a long time and couldn't find a solution, I manually wrote the location of the fiducials (in map.txt) and then used fiducial_slam to locate the robot. But the Odom and the base_footprint in rviz jittered and even far above the environment of the map.
[image]
Any help is greatly appreciated!!
Launch file I used: 
```
<launch>

  <arg name="fiducial_len" default="0.14"/>
  <arg name="dictionary" default="7"/>
  <arg name="do_pose_estimation" default="true"/>

  <arg name="vis_msgs" default="false"/>
  <arg name="ignore_fiducials" default="" />
  <arg name="fiducial_len_override" default="" />
  <arg name="verbose" default="false"/>

  <arg name="camera" default="/test_camera"/>
  <arg name="map_frame" default="map"/>
  <arg name="odom_frame" default="odom"/>
  <arg name="base_frame" default="base_footprint"/>
  <arg name="publish_tf" default="true"/>
  <arg name="tf_publish_interval" default="2.0"/>
  <arg name="future_date_transforms" default="0.1"/>
  <arg name="publish_6dof_pose" default="true"/>
  <arg name="pose_publish_rate" default="20"/>
  <arg name="systematic_error" default="0.01"/>
  <arg name="covariance_diagonal" default=""/>

  <node pkg="aruco_detect" name="aruco_detect" type="aruco_detect" output="screen" respawn="false">
    <param name="publish_images" value="true" />
    <param name="fiducial_len" value="$(arg fiducial_len)"/>
    <param name="dictionary" value="$(arg dictionary)"/>
    <param name="do_pose_estimation" value="$(arg do_pose_estimation)"/>
    <param name="vis_msgs" value="$(arg vis_msgs)"/>
    <param name="ignore_fiducials" value="$(arg ignore_fiducials)"/>
    <param name="fiducial_len_override" value="$(arg fiducial_len_override)"/>
    <param name="verbose" value="$(arg verbose)"/>
    <remap from="camera" to="/test_camera/image_raw"/>
    <remap from="camera_info" to="/camera_info"/>
  </node>
</launch>
```
```
<launch>    
    <arg name="camera" default="/test_camera"/>
  <arg name="map_frame" default="map"/>
  <arg name="odom_frame" default="odom"/>
  <arg name="base_frame" default="base_footprint"/>
  <arg name="publish_tf" default="true"/>
  <arg name="tf_publish_interval" default="0.2"/>
  <arg name="future_date_transforms" default="0.0"/>
  <arg name="publish_6dof_pose" default="false"/>
  <arg name="pose_publish_rate" default="20"/>
  <arg name="systematic_error" default="0.01"/>
  <arg name="covariance_diagonal" default="[0.0, 0.0, 0.0, 0.0, 0.0, 0.0]"/>
  <node type="fiducial_slam" pkg="fiducial_slam" output="screen" name="fiducial_slam">
    <param name="map_file" value="/home/dwa/catkin_evo/src/kion_sim/kion_sim_bringup/cfg/map.txt" />
    <param name="map_frame" value="$(arg map_frame)" />
    <param name="odom_frame" value="$(arg odom_frame)" />
    <param name="base_frame" value="$(arg base_frame)" />
    <param name="publish_tf" value="$(arg publish_tf)" />
    <param name="tf_publish_interval" value="$(arg tf_publish_interval)" />
    <param name="future_date_transforms" value="$(arg future_date_transforms)" />
    <param name="publish_6dof_pose" value="$(arg publish_6dof_pose)" />
    <param name="pose_publish_rate" value="$(arg pose_publish_rate)" />
    <param name="sum_error_in_quadrature" value="true"/>
    <param name="read_only_map" value="false"/>
    <rosparam param="covariance_diagonal" subst_value="True">$(arg covariance_diagonal)</rosparam>
    <remap from="/camera_info" to="/camera_info"/>
  </node>
</launch>
 ```

MoffKalast, 2022-02-27:
> This package implements a system that uses ceiling mounted fiducial markers (think QR Codes) to allow a robot to identify its location and orientation.

> a system that uses ceiling mounted fiducial markers 

> ceiling mounted

There's your problem.

Wayne-xixi, 2022-03-03:
> > This package implements a system that uses ceiling mounted fiducial markers (think QR Codes) to allow a robot to identify its location and orientation.
> 
> > a system that uses ceiling mounted fiducial markers
> 
> > ceiling mounted
> 
> There's your problem. The system makes the assumption that all the markers are on the same horizontal plane above the robot, and it won't really work for your setup with any reasonable accuracy.

Thanks

closed 2022-03-03 by Wayne-xixi