UbiquityRobotics/fiducials #271: Exact orientation of tf for Fiducial

pitosalas, 2022-04-01:
I'm troubleshooting a problem with my use of Armco_detect. It seems to work well except that the fiducials tfs relatvive to the camera are 90 degrees in the wrong direction. The fiducial_imates topic shows the fiducials. But in stead of x-y-z colored lines to show the orientation, I see the following

- a green outline around the marker
- a red square in the top left
- a blue id = x label.

Can that be changed into a x,y,z axis in rgb? That would be much more useful

anfederman, 2022-04-01:
Is the camera mounted to a robot, and is there a link to the base_link or
base_footprint? I am guessing the camera is oriented incorrectly.

pitosalas, 2022-04-01:
Yes, there is a link. But the transform is wrong. 

My question really was about the visualization (fiducial_camera I think it's called). The image shows the outline and number of the fiducial. But more helpful would be if I could get it to show the three axes in perspective (like it does in the demo visualization). Seeing the three axes will really help me troubleshoot!

Pito Salas
Faculty, Computer Science
Brandeis University

JanezCim, 2022-04-06:
@pitosalas might be a bit of workaround, but you can see the three axis prespective if you enable tf visualisation in rviz. If you do fiducial slam, that also publishes markers of fiducials: 

[image]

but afaik this node does not have an option of displaying frame coordiante system in the image stream.

ivan140, 2022-09-02:
You can add the following code to map.cpp , function publishMarker :

```cpp
// Arrow
visualization_msgs::Marker arrow;
arrow.type = visualization_msgs::Marker::ARROW;
arrow.action = visualization_msgs::Marker::ADD;
toMsg(fid.pose.transform, arrow.pose);

arrow.scale.x = 0.15;
arrow.scale.y = 0.05;
arrow.scale.z = 0.05;
if (fid.visible) {
    arrow.color.r = 1.0f;
    arrow.color.g = 0.0f;
    arrow.color.b = 0.0f;
    arrow.color.a = 1.0f;
} else {
    arrow.color.r = 1.0f;
    arrow.color.g = 0.0f;
    arrow.color.b = 0.0f;
    arrow.color.a = 1.0f;
}
tf2::Quaternion q = fid.pose.transform.getRotation();

arrow.pose.orientation.x = q.getX();
arrow.pose.orientation.y = q.getY();
arrow.pose.orientation.z = q.getZ();
arrow.pose.orientation.w = q.getW();
arrow.id = fid.id;
arrow.ns = "fiducial";
arrow.header.frame_id = mapFrame;
markerPub.publish(arrow);
```

And get the following output. Feel free to add more axis if needed.
    
[image]

pitosalas, 2022-10-11:
Thanks!

Pito Salas
Faculty, Computer Science
Brandeis University

closed 2022-12-01 by anfederman