UbiquityRobotics/fiducials #279: TF problem when I detect the aruco markers

wowowolala, 2022-11-30:
I run the aruco_detect.launch, but the tf about camera and aruco marker is wrong, like picture below.
[image]

The tf should be like this:  (for example)
[image]

How can I modify the tf to be correct?

wowowolala, 2022-11-30:
It is not camera problem.

anfederman, 2022-11-30:
Do you have the camera orientation correct?

anfederman, 2022-11-30:
This is the same issue as 271 and 275.   Perhaps reying STAG makers instead will resolve, as I don't think we have any LTS for aruco_detect.

anfederman, 2022-11-30:
OK  I just tried STAG markers in our easymap stack.   This on a Magni robot with the camera forward.   It still puts the maker below the floor. even when it is on the wall directly ahead of the robot.

[image]

[image]

anfederman, 2022-11-30:
Since the TF error is in both the magni and easymap stacks, It must have something to do with camera orientation.  It appears the transform thinks the camera is in the tower orientation rather than the lower magni forward orientation. This would explain why rviz put the maker below the floor.  I have tried changing the camera orientation in the URDF section but nothing seem to fix the issue.

we need documentation and pictures  CAREFULLY explaining and showing the camera installed and the  extrinsics config files.   SOmewhere in the history of going from magni to breadcrumb, this information has been lost.

wowowolala, 2022-12-01:
> Do you have the camera orientation correct?
> […](#)

yes, because I use same camera file to detect apriltag and aruco marker.
The result of detecting apriltag is correct like picture 2.

wowowolala, 2022-12-01:
This is my rs_camera.launch

<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="camera"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="15"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="false"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="false"/>

[... 57 lines cut ...]

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="reconnect_timeout"        value="$(arg reconnect_timeout)"/>
      <arg name="wait_for_device_timeout"  value="$(arg wait_for_device_timeout)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>

      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
    </include>
  </group>
</launch>

wowowolala, 2022-12-01:
And this is my aruco_detect.launch

<launch>
  
  <arg name="camera" default="/camera/color"/>
  <arg name="image" default="image_raw"/>
  <arg name="transport" default="compressed"/>
  <arg name="fiducial_len" default="0.14"/>
  <arg name="dictionary" default="7"/>
  <arg name="do_pose_estimation" default="true"/>
  
  <arg name="vis_msgs" default="false"/>
  <arg name="ignore_fiducials" default="" />
  <arg name="fiducial_len_override" default="10: 0.1, 11: 0.1, 20: 0.04" />
  <arg name="verbose" default="false"/>

  <node pkg="aruco_detect" name="aruco_detect"
    type="aruco_detect" output="screen" respawn="false">
    <param name="image_transport" value="$(arg transport)"/>
    <param name="publish_images" value="true" />
    <param name="fiducial_len" value="$(arg fiducial_len)"/>
    <param name="dictionary" value="$(arg dictionary)"/>
    <param name="do_pose_estimation" value="$(arg do_pose_estimation)"/>
    <param name="vis_msgs" value="$(arg vis_msgs)"/>
    <param name="ignore_fiducials" value="$(arg ignore_fiducials)"/>
    <param name="fiducial_len_override" value="$(arg fiducial_len_override)"/>
    <param name="verbose" value="$(arg verbose)"/>
    <remap from="camera/compressed" 
        to="$(arg camera)/$(arg image)/$(arg transport)"/>
    <remap from="camera_info" to="$(arg camera)/camera_info"/>
  </node>
</launch>

wowowolala, 2022-12-01:
> Since the TF error is in both the magni and easymap stacks, It must have something to do with camera orientation. It appears the transform thinks the camera is in the tower orientation rather than [...]

Same situation! I just run rs_camera.launch and aruco_detect.launch.
and I found it think that aruco marker is on z axis of camera!
But the correct is aruco marker should be on x axis of camera.

anfederman, 2022-12-01:
[image]. 

I have it working.   

the camera needs to be mounted with the cable coming out the bottom, ie toward the PI.   

 Have the camera position specified in the robot.yaml

# Check for lidar and camera extrinsics files in two places with following prior
ities:

#  1.) in ~/.ros/extrinsics/<SENSOR>_extrinsics_<POSITION>.yaml
#  2.) in package magni_description/extrinsics/<SENSOR>_extrinsics_<POSITION>.yaml
# where <SENSOR> is either "camera" or "lidar" and <POSITION> is an arbitrary string with which the extrinsics file is identified

raspicam_position: 'forward' # to disable insert "None"
lidar_position : 'top_plate'  # to disable insert "None"

good luck

closed 2022-12-01 by anfederman